sparkfun/piservohat_py #4: Safe pwm changes between pulses but never during pulses

Petrox, 2020-10-29:
Care should be taken that changes in the PWM width  (setting the length) is only possible during the low state. (if the pulse width was 1500usec, and I set it to 0usec, but this would happen when the signal is high, then I'd cut the last pulse to a random shorter length pulse, that could be dangerous and cause hardware failure (eg a random 900usec pulse before going dark could break a digital servo, especially if it is installed in some machine and the motion range is limited)  

Please confirm (in the docs), that a code that toggles the pwm between 70deg to 170deg 10000000 times there will be not a single event when the pulse is interrupted at a random length and the servo won't ever receive any commands that are out of the 70-170 deg range. 

To clarify: If I change between 1200 and 1600 microsec often then I should never ever see a single pulse that is shorter than 1200 microsec. Not even if I set the pulse width to 600 microsec and then 1200 and back, I should never see no 300 microsec pulses nor 900 microsec pulses nor 1500 microsec pulses on a scope.

Please confirm that this has been handled correctly and the hat is not dangerous to digital servos and any other electronics depending on the correct way of changing pwm signal length.

santaimpersonator, 2021-02-18:
Are you sure this is an issue? I am 98% sure that the hardware won't do that, based on its operation as specified in the [datasheet](https://www.nxp.com/docs/en/data-sheet/PCA9685.pdf). However, I have worked on several other products since, so I could be wrong. Again, I'd have to have an opportunity to look at this; I am in the middle of several things.

santaimpersonator, 2021-07-01:
Not possible: Please see code and datasheet.

Closing because issue is invalid.

closed 2021-07-01 by santaimpersonator