sparkfun/piservohat_py #5: float/int conversion issue

Petrox, 2020-10-29:
After fixing the prescaler float issue in the qwiic_pca9685 there are similar issues still remaining:
````
    self.mySensor.set_duty_cycle(ch, 0)
  File "/usr/local/lib/python2.7/dist-packages/pi_servo_hat.py", line 376, in set_duty_cycle
    self.PCA9685.set_channel_word(channel, 1, on_value)	# Timing for "On" edge of PWM
  File "build/bdist.linux-armv7l/egg/qwiic_pca9685.py", line 1117, in set_channel_word
  File "/usr/local/lib/python2.7/dist-packages/sparkfun_qwiic_i2c-0.9.11-py2.7.egg/qwiic_i2c/linux_i2c.py", line 221, in writeWord
    return self.i2cbus.write_word_data(address, commandCode, value)
  File "build/bdist.linux-armv7l/egg/smbus2/smbus2.py", line 474, in write_word_data
TypeError: int expected instead of float
````

environment was set up in docker container (ros:melodic) on a raspbian raspberry 4.

```
Step 6/7 : RUN cd pypi/tempfixforfloat/Qwiic_PCA9685_Py-patch-1 ; python setup.py install ; python -m pip install sparkfun-pi-servo-hat RPi.GPIO
 ---> Running in ff08171ebd79
running install
running bdist_egg
running egg_info
writing requirements to sparkfun_qwiic_pca9685.egg-info/requires.txt
writing sparkfun_qwiic_pca9685.egg-info/PKG-INFO
writing top-level names to sparkfun_qwiic_pca9685.egg-info/top_level.txt
writing dependency_links to sparkfun_qwiic_pca9685.egg-info/dependency_links.txt
reading manifest file 'sparkfun_qwiic_pca9685.egg-info/SOURCES.txt'
writing manifest file 'sparkfun_qwiic_pca9685.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build/bdist.linux-armv7l/egg
copying build/lib.linux-armv7l-2.7/qwiic_pca9685.py -> build/bdist.linux-armv7l/egg
byte-compiling build/bdist.linux-armv7l/egg/qwiic_pca9685.py to qwiic_pca9685.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying sparkfun_qwiic_pca9685.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying sparkfun_qwiic_pca9685.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying sparkfun_qwiic_pca9685.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying sparkfun_qwiic_pca9685.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying sparkfun_qwiic_pca9685.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating 'dist/sparkfun_qwiic_pca9685-0.9.1-py2.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing sparkfun_qwiic_pca9685-0.9.1-py2.7.egg
Copying sparkfun_qwiic_pca9685-0.9.1-py2.7.egg to /usr/local/lib/python2.7/dist-packages
Adding sparkfun-qwiic-pca9685 0.9.1 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/sparkfun_qwiic_pca9685-0.9.1-py2.7.egg
Processing dependencies for sparkfun-qwiic-pca9685==0.9.1
Searching for sparkfun_qwiic_i2c
Reading https://pypi.python.org/simple/sparkfun_qwiic_i2c/
Downloading https://files.pythonhosted.org/packages/f4/6d/c474b9b8963654363e1b232eb0a3f85432f88f7e67987803801df0bfe647/sparkfun_qwiic_i2c-0.9.11-py2.py3-none-any.whl#sha256=0978cd133c8e86fece2c85122a6a2e6ab043fce5ed29d348dfe6dfb758b09972
Best match: sparkfun-qwiic-i2c 0.9.11
Processing sparkfun_qwiic_i2c-0.9.11-py2.py3-none-any.whl
Installing sparkfun_qwiic_i2c-0.9.11-py2.py3-none-any.whl to /usr/local/lib/python2.7/dist-packages
writing requirements to /usr/local/lib/python2.7/dist-packages/sparkfun_qwiic_i2c-0.9.11-py2.7.egg/EGG-INFO/requires.txt
Adding sparkfun-qwiic-i2c 0.9.11 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/sparkfun_qwiic_i2c-0.9.11-py2.7.egg
Searching for smbus2
Reading https://pypi.python.org/simple/smbus2/
Downloading https://files.pythonhosted.org/packages/6a/06/80a6928e5cbfd40c77c08e06ae9975c2a50109586ce66435bd8166ce6bb3/smbus2-0.3.0.tar.gz#sha256=210e66eebe4d0b1fe836b3ec2751841942e1c4918c0b429b20a0e20a222228b4
Best match: smbus2 0.3.0
Processing smbus2-0.3.0.tar.gz
Writing /tmp/easy_install-Qzc0yv/smbus2-0.3.0/setup.cfg
Running smbus2-0.3.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-Qzc0yv/smbus2-0.3.0/egg-dist-tmp-4bgVBa
zip_safe flag not set; analyzing archive contents...
Moving smbus2-0.3.0-py2.7.egg to /usr/local/lib/python2.7/dist-packages
Adding smbus2 0.3.0 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/smbus2-0.3.0-py2.7.egg
Finished processing dependencies for sparkfun-qwiic-pca9685==0.9.1
Collecting sparkfun-pi-servo-hat
  Downloading https://files.pythonhosted.org/packages/c3/6e/9283ac95df4fe57940c0fd772aa6f82a7b2d5e3242dc350f34fff418745e/sparkfun_pi_servo_hat-0.9.0-py2.py3-none-any.whl
Collecting RPi.GPIO
  Downloading https://files.pythonhosted.org/packages/cb/88/d3817eb11fc77a8d9a63abeab8fe303266b1e3b85e2952238f0da43fed4e/RPi.GPIO-0.7.0.tar.gz
Requirement already satisfied: sparkfun-qwiic-pca9685 in /pypi/tempfixforfloat/Qwiic_PCA9685_Py-patch-1 (from sparkfun-pi-servo-hat)
Requirement already satisfied: sparkfun-qwiic-i2c in /usr/local/lib/python2.7/dist-packages/sparkfun_qwiic_i2c-0.9.11-py2.7.egg (from sparkfun-pi-servo-hat)
Requirement already satisfied: smbus2 in /usr/local/lib/python2.7/dist-packages/smbus2-0.3.0-py2.7.egg (from sparkfun-qwiic-i2c->sparkfun-pi-servo-hat)
```

santaimpersonator, 2021-02-19:
As mentioned here (https://github.com/sparkfun/Qwiic_PCA9685_Py/pull/1#issuecomment-781709751), probably a Python 2 vs Python 3 thing. I probably just have to go around and update that.

santaimpersonator, 2021-07-01:
The official word is we aren't supporting Python 2 because it has been [retired](https://www.python.org/psf/press-release/pr20191220/).

I am going to close this issue as it is redundant, since the `round()` function already provides an integer (in Python 3).

closed 2021-07-01 by santaimpersonator