sparkfun/piservohat_py #6: Continuous Rotation Servo Support

ashields1965, 2022-01-14:
Does the hardware/Library support Continuous Rotation Servos? Can not find any references In the API reference

Many thanks..

fab1an, 2024-01-14:
I would like to know the same.
@santaimpersonator Could you answer this?

sfe-SparkFro, 2024-01-15:
FWIW continuous rotation servos use the exact same PWM signal as normal servos, so you could use this library to control both types of servos even if there aren't functions explicitly named for CR servos.

To elaborate a bit, normal servos typically use 1ms and 2ms pulses to define the ends of their range, with a 1.5ms pulse corresponding to the center position. For CR servos, they typically use a 1.5ms pulse to define zero rotation speed, and 1ms and 2ms pulses to define max forwards and backwards rotation speed.

With that in mind, this should be sufficient to make a CR servo spin:

```
# Servo channel to use
channel = 0

# Full backwards (using default "90 degree servo")
my_servo_hat.move_servo_position(channel , 0)

# Stop (using default "90 degree servo")
my_servo_hat.move_servo_position(channel, 45)

# Full forwards (using default "90 degree servo")
my_servo_hat.move_servo_position(channel, 90)
```

Or if you find the 90 degree thing confusing, you could also do something like this:

```
# Servo channel to use
channel = 0

# Max value for servo control (0 is always the bottom limit)
max_value = 1

# Full backwards
my_servo_hat.move_servo_position(channel , 0, max_value)

# Stop
my_servo_hat.move_servo_position(channel, 0.5, max_value)

# Full forwards
my_servo_hat.move_servo_position(channel, 1, max_value)
```

fab1an, 2024-01-16:
Thank you for the write-up!

fab1an, 2024-01-29:
> Or if you find the 90 degree thing confusing, you could also do something like this:
> 
> ```
> # Servo channel to use
> channel = 0
> 
> # Max value for servo control (0 is always the bottom limit)
> max_value = 1
> 
> # Full backwards
> my_servo_hat.move_servo_position(channel , 0, max_value)

Just a heads-up, this is not correct, as the code for `move_servo_position` only allows a third parameter of `90` or `180`: https://github.com/sparkfun/PiServoHat_Py/blob/main/pi_servo_hat.py#L331.

sfe-SparkFro, 2024-06-28:
Closing this issue since CR servos can be controlled with this library. It is admittedly a bit clunky, but it works at least.

closed 2024-06-28 by sfe-SparkFro